dji-sdk/Onboard-SDK-ROS #526: A3pro support for 3.8.1

lianghanfang, 2023-07-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35826'>Zendesk ticket #35826</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for any inconvenience with version update, The N3 supports ROS 3.8.
The following log shows some wrong with advance sensing,  The N3 not support  advance sensing.
so you need to switch off the ADVANCED_SENSING function and modify the code dji_sdk_node.cpp as follows:
 [image]

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

_Originally posted by @dji-dev in https://github.com/dji-sdk/Onboard-SDK-ROS/issues/348#issuecomment-644658006_

'''
roslaunch dji_sdk sdk.launch 
... logging to /home/lhf/.ros/log/4ecd206a-22e7-11ee-ae42-2d9844f14063/roslaunch-lhf-NUC11TNKv5-18114.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lhf-NUC11TNKv5:45457/

SUMMARY
========

PARAMETERS
 * /dji_sdk/acm_name: /dev/ttyACM0
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1134844
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: 38ce3b1aa85a5620c...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [18122]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4ecd206a-22e7-11ee-ae42-2d9844f14063
process[rosout-1]: started with pid [18132]
started core service [/rosout]
process[dji_sdk-2]: started with pid [18139]

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB0 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = 0670150144

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = PM820V3PRO

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.2.41.14

ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1313: version 0x302290E

STATUS/1 @ activate, L1326: Activation successful

STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ initGimbal, L890: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ initGimbal, L934: Gimbal not mounted!
[ INFO] [1689408741.833545515]: drone activated
[dji_sdk-2] process has died [pid 18139, exit code -11, cmd /home/lhf/osdk_ws/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/lhf/.ros/log/4ecd206a-22e7-11ee-ae42-2d9844f14063/dji_sdk-2.log].
log file: /home/lhf/.ros/log/4ecd206a-22e7-11ee-ae42-2d9844f14063/dji_sdk-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
'''

lianghanfang, 2023-07-15:
我按照上面的图把advanced sensing宏定义中true改成了false，但是底下还是出现了错误
'''
[dji_sdk-2] process has died [pid 18139, exit code -11, cmd /home/lhf/osdk_ws/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/lhf/.ros/log/4ecd206a-22e7-11ee-ae42-2d9844f14063/dji_sdk-2.log].
log file: /home/lhf/.ros/log/4ecd206a-22e7-11ee-ae42-2d9844f14063/dji_sdk-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
'''
不清楚是什么原因，在把关于advanced sensing中所有的宏都屏蔽掉后还是会有问题

dji-dev, 2023-07-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/86968'>Zendesk ticket #86968</a>:<br><br>可以用GDB跑一下看崩溃在什么地方。或者直接先不修改sample编译运行看是否可以。
<br><p><em><small>°°°</small></em></p>