dji-sdk/Onboard-SDK-ROS #529: DJI M300 DRONE IMU & CAMERA COORDINATE ?

mahammadirfan, 2023-09-07:
Hi,
We have a matrice 300 drone and i am using the stereo camera and imu both with dji-osdk-ros......my question is what is the coodinate system WHICH is fixed for IMU & cameras by DJI ?? How can i know both the coordinates ?

 I have one more question.....in m300 drone can i able to start all the stereo pair cameras simultaneously same time for pointcloud ??

Thanks!

dji-dev, 2023-09-11:
Agent comment from shiyao.liao in <a href='https://djisdksupport.zendesk.com/agent/tickets/90605'>Zendesk ticket #90605</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Regarding your first question, I will transfer it to the relevant team to answer your question, and they will reply to you soon.
Regarding your second question, we are sorry, but we do not support all cameras to perform point clouds at the same time.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>

mahammadirfan, 2023-12-12:
Hi,
I am still waiting for the answer of my question i.e the cordinate system of the IMU and does dji has transformation matrix for the imu to camera of m300 ?
Thanks in advance!

dji-dev, 2023-12-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/90605'>Zendesk ticket #90605</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

IMU is the FLU body coordinate system. Regarding camera coordinates, our code package contains simple models of open aircraft and cameras. It is recommended that you import the modeling software and measure the data yourself to offset and calculate the coordinates.

Thank you for your support of DJI products.

Best Regards,
DJI Dajiang innovation SDK technical support
<br><p><em><small>°°°</small></em></p>