dji-sdk/Onboard-SDK-ROS #534: Not able to publish to the topic: /dji_sdk/flight_control_setpoint_rollpitch_yawrate_zposition

adeelahsaanawan, 2024-04-30:
My final aim to design a NMPC controller that would generate a pitch, roll angles, yaw rate commands, and z position command.

What I am trying to do is to publish to the following topic:

`/dji_sdk/flight_control_setpoint_rollpitch_yawrate_zposition`

To do this, I first launch the `dji_sdk_node`. Then, I open another terminal, and run the following command in terminal:

`rostopic pub /dji_sdk/flight_control_setpoint_rollpitch_yawrate_zposition sensor_msgs/Joy "{header: {stamp: {secs: 0, nsecs: 0}, frame_id: ''}, axes: [0.0, 0.0, 0.0, 0.5], buttons: []}"`

When i run the command, it says "publishing and latching message", but I don't see anything happening in the DJI Assitant 2 Simulator. 

But, When I publish to the `/dji_sdk/flight_control_setpoint_generic`, it works and I see the drone moving in the simulator. I think it has to do something with the flags. Could anyone please help me out here!

adeelahsaanawan, 2024-05-03:
Thanks for getting back to me, I await your response.

dji-dev, 2024-05-06:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/106373'>Zendesk ticket #106373</a>:<br><br>Dear adeelahsaanawan,

Greetings! We appreciate your contact with DJI - Da-Jiang Innovations.

Thank you for your patience. May we inquire about the specific version of ROS you are using and the model of your drone? We may need to verify this information first before we can further assist you.

Thank you for your support of DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

adeelahsaanawan, 2024-05-10:
I am using ROS Kinetic, and DJI M300. But I have resolved the issue, I was not obtaining the control authority.

dji-dev, 2024-05-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/106373'>Zendesk ticket #106373</a>:<br><br>Dear adeelahsaanawan,

Greetings. We appreciate your contact with DJI - Da-Jiang Innovations.

We apologize for any inconvenience caused by the delay. We are pleased to hear that you have resolved the issue. Should you have any further requests, please do not hesitate to contact us again. Thank you once more.

We appreciate your support for DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>