dji-sdk/Onboard-SDK-ROS #535: Can not access H20T camera data via OSDK on iCrest

Erfan-Dilfanian, 2024-05-13:
Hello,
I am using DJI M300 RTK with iCrest 2.0 as an onboarding computer, and H20T camera
I can not access camera images via iCrest's OSDK
For example, after I launch dji_vehicle_node.launch, and then when I run main_and_fpv_node node, I can not access images and the following is the error that I will get in the roscore terminal:
STATUS/1All components for decoding initialized ...

STATUS/1User callback thread created successfully!

STATUS/1****** Decoder Callback Thread Start ******

[1943509.053]STATUS/1 @ getCameraList, L218: Request start pushing camera info ack = 0

[1943510.074]STATUS/1 @ getCameraList, L235: [1943510.074]STATUS/1 @ getCameraList, L236: mounted	cam0:1 cam2:0 cam3:0
[1943510.074]STATUS/1 @ getCameraList, L237: Decoder Callback Thread: Get image time out

type   	cam0:43 cam2:0 cam3:0

Or sometimes the following error:

STATUS/1User callback thread created successfully!

STATUS/1****** Decoder Callback Thread Start ******

[1944168.711]STATUS/1 @ getCameraList, L218: Request start pushing camera info ack = 0

[1944169.730]STATUS/1 @ getCameraList, L235: [1944169.730]STATUS/1 @ getCameraList, L236: [1944169.730]STATUS/1 @ getCameraList, L237: [1944169.730]STATUS/1 @ startH264Stream, L456: [1944169.732]ERRORLOG/1 @ subscribeLiveViewData, L323: [1944169.732]ERRORLOG/1 @ subscribeLiveViewData, L325: Subscribe raw error code : 0x89

Erfan-Dilfanian, 2024-05-27:
Hello again
In a terminal, I write down the following command to launch the launch file:
roslaunch dji_osdk_ros dji_vehicle_node.launch
And after the activation of the launch file in the first terminal was successful, I run the following command on another terminal to access main camera info:
rosrun dji_osdk_ros main_and_fpv_node
The following is the complete log of the first terminal:
... logging to /home/wu/.ros/log/6cc3fcc0-1c55-11ef-be4d-54ef33fc3869/roslaunch-ubuntu-6291.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://172.31.16.194:37561/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1106264
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 1000000
 * /vehicle_node/enc_key: dd694ebc9089de602...
 * /vehicle_node/serial_name: /dev/ttyTHS1
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [6300]
ROS_MASTER_URI=http://172.31.16.194:11311

setting /run_id to 6cc3fcc0-1c55-11ef-be4d-54ef33fc3869
process[rosout-1]: started with pid [6310]
started core service [/rosout]
process[vehicle_node-2]: started with pid [6313]
EnableAd: 1
[3141921.173]STATUS/1 @ getDroneVersion, L1702: ret = 0
[3141921.173]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNBJ4500C00B6
[3141921.173]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.18.29
[3141921.173]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3141921.174]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3141922.174]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[3141922.374]STATUS/1 @ Control, L40: The control class is going to be deprecated.It will be better to use the FlightController class instead!
[3141922.374]STATUS/1 @ FileMgrImpl, L254: register download file callback handler successfully.
[3141923.644]STATUS/1 @ Firewall, L45: Firewall is initializing ...
[3141923.644]STATUS/1 @ Firewall, L65: osdk policy file updating(1) ......
[3141924.646]STATUS/1 @ Firewall, L73: osdk policy file updating(2) ......
[3141924.648]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:0
[3141924.648]STATUS/1 @ RequestUploadPolicyFileHandle, L234: Upload policy file info md5 checksum and version
[3141924.649]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[3141924.649]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 0 0 200
[3141924.650]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:1
[3141924.650]STATUS/1 @ RequestUploadPolicyFileHandle, L254: request upload policy file data: 1 200 175
[3141924.653]STATUS/1 @ RequestUploadPolicyFileHandle, L217: request upload policy file type:2
[3141924.653]STATUS/1 @ RequestUploadPolicyFileHandle, L282: request upload policy file success
[3141925.646]STATUS/1 @ firewallTask, L112: firewall task created ...
[3141925.653]STATUS/1 @ getDroneVersion, L1702: ret = 0
[3141925.653]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNBJ4500C00B6
[3141925.653]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.18.29
[3141925.693]ERRORLOG/1 @ AdvancedSensing, L108: Failed to initialize USB Bulk Linker channel for liveview!
[3141925.693]ERRORLOG/1 @ AdvancedSensing, L115: Failed to create task for liveview!
[3141925.695]ERRORLOG/1 @ AdvancedSensing, L123: Failed to initialize USB Bulk Linker channel for perception!
[3141925.695]ERRORLOG/1 @ AdvancedSensing, L130: Failed to create task for advanced sensing!
[3141925.695]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[3141925.695]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[... 23 lines cut ...]

[3141927.815]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[3141927.815]STATUS/1 @ startPackage, L347: Package 4 info: freq=400, nTopics=1.

STATUS/1All components for decoding initialized ...

STATUS/1User callback thread created successfully!

STATUS/1****** Decoder Callback Thread Start ******

[3141954.873]STATUS/1 @ getCameraList, L218: Request start pushing camera info ack = 0

[3141955.893]STATUS/1 @ getCameraList, L235: [3141955.893]STATUS/1 @ getCameraList, L236: [3141955.893]STATUS/1 @ getCameraList, L237: [3141955.893]STATUS/1 @ startH264Stream, L456: camera[0] is mounted

Decoder Callback Thread: Get image time out

[3141955.895]STATUS/1 @ subscribeLiveViewData, L321: subsrcibe data success!

Decoder Callback Thread: Get image time out

And the following is contents of the second terminal:
Please input the camera type you want to subscribe: 0 for FPV_CAM, 1 for MAIN_CAM
1

I can not access image data from H20T.

Erfan-Dilfanian, 2024-05-27:
[image]

dji-dev, 2024-05-28:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/107245'>Zendesk ticket #107245</a>:<br><br>Dear Erfan-Dilfanian,

Greetings. Thank you for reaching out to DJI - Da-Jiang Innovations.

Failed to initialize USB Bulk Linker channel for liveview!
[3141925.693]ERRORLOG/1 @ AdvancedSensing, L115: Failed to create task for liveview!
[3141925.695]ERRORLOG/1 @ AdvancedSensing, L123: Failed to initialize USB Bulk Linker channel for perception!
[3141925.695]ERRORLOG/1 @ AdvancedSensing, L130: Failed to create task for advanced sensing!

We have identified the error in the information you provided. It appears that the USB bulk link was not initialized properly. We recommend checking the connection of the USB cable.

Thank you for your support of DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>

Erfan-Dilfanian, 2024-05-29:
Hi,
What USB do you mean? The iCrest2 is connected to M300 with a cable which does not look like USB to me.

dji-dev, 2024-05-29:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/107245'>Zendesk ticket #107245</a>:<br><br>Dear Erfan-Dilfanian,

Greetings. We appreciate your contact with DJI - Da-Jiang Innovations.

Could you please refer to the connection method shown in the attached image? The parts marked in the image need to be replaced with a TypeC cable, and then the DIP switch on the E-Port expansion board should be set to the Device side. There are two hardware connections in total, one is a serial port, and the other is USB.
https://developer.dji.com/doc/payload-sdk-tutorial/en/quick-start/quick-guide/jetson-nano.html

Thank you for your support of DJI products! We wish you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support<br><a href='https://djisdksupport.zendesk.com/attachments/token/fdoaydUKK9f4CV46UtGtERF7f' target='zd_attach'>image.png</a>
<br><p><em><small>°°°</small></em></p>

Erfan-Dilfanian, 2024-05-29:
i seemingly skipped the error using the following three lines:
sudo chmod -R +rwx /dev/ttyTHS1
sudo chmod +rwx -R /dev/ttyACM0
sudo usermod -a -G dialout $USER
And then restarted and typed the following:
sudo vim /etc/udev/rules.d/DJIDevice.rules
SUBSYSTEM=="usb", ATTRS{idVendor}=="2ca3", MODE ="0666"
However, when I wanna rotate the gimbal, I have the following erors in the launch file terminal:
[ INFO] [1717023064.078821471]: Obtain Control Authority Callback
[ INFO] [1717023070.996925403]: call takeoff service
Motors spinning...
Ascending...
Successful takeoff!
[ INFO] [1717023076.313479901]: call move local position offset service
[ INFO] [1717023087.045442415]: call move local position offset service
[ INFO] [1717023094.006554346]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 3.98)
[ INFO] [1717023094.006787597]: Call gimbal Ctrl.
[3331142.902]STATUS/1 @ rotateGimbal, L1902: Start to rotate the gimbal 0, (p,r,y) = (-30.00,0.00,0.00) mode:0 time:0.00s
[ INFO] [1717023096.041416535]: Current gimbal 0 , angle (p,r,y) = (0.00, -29.90, 3.98)
[ INFO] [1717023104.588948471]: Current gimbal 0, angle (p,r,y) = (0.00, -29.90, 3.98)
[ INFO] [1717023104.589178010]: Call gimbal Ctrl.
[3331153.485]STATUS/1 @ rotateGimbal, L1902: Start to rotate the gimbal 0, (p,r,y) = (-50.00,0.00,0.00) mode:0 time:1.00s
[ INFO] [1717023106.620871017]: Current gimbal 0 , angle (p,r,y) = (0.00, -79.90, 3.98)
[ INFO] [1717023106.926845939]: Current gimbal 0, angle (p,r,y) = (0.00, -79.90, 3.98)
[ INFO] [1717023106.927061557]: Call gimbal Ctrl.
[3331155.823]STATUS/1 @ rotateGimbal, L1902: Start to rotate the gimbal 0, (p,r,y) = (-15.00,0.00,0.00) mode:0 time:2.50s
[ INFO] [1717023108.960965698]: Current gimbal 0 , angle (p,r,y) = (0.00, -91.70, 3.98)
[ INFO] [1717023109.266763658]: Current gimbal 0, angle (p,r,y) = (0.00, -93.60, 3.98)
[ INFO] [1717023109.266972428]: Call gimbal Ctrl.
[3331158.163]STATUS/1 @ rotateGimbal, L1902: Start to rotate the gimbal 0, (p,r,y) = (-30.00,0.00,0.00) mode:0 time:1.50s
[3331158.197]ERRORLOG/1 @ rotateGimbal, L1909: Rotation gimbal fail. Error code : 0xC00000000FF
[3331158.197]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Gimbal
[3331158.197]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Undefined error
[3331158.197]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Please contact <dev@dji.com> for help.
[ INFO] [1717023111.302058122]: Current gimbal 0 , angle (p,r,y) = (0.00, -119.90, 3.98)
[ INFO] [1717023120.148850777]: Current gimbal 0, angle (p,r,y) = (0.00, -119.90, 3.98)
[ INFO] [1717023120.149070907]: Call gimbal Ctrl.
[3331169.045]STATUS/1 @ rotateGimbal, L1902: Start to rotate the gimbal 0, (p,r,y) = (-50.00,0.00,0.00) mode:0 time:1.00s
[3331169.077]ERRORLOG/1 @ rotateGimbal, L1909: Rotation gimbal fail. Error code : 0xC00000000FF
[3331169.077]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Gimbal
[3331169.077]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Undefined error
[3331169.077]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Please contact <dev@dji.com> for help.
[ INFO] [1717023122.182244106]: Current gimbal 0 , angle (p,r,y) = (0.00, -120.00, 3.98)
[ INFO] [1717023122.487895186]: Current gimbal 0, angle (p,r,y) = (0.00, -119.90, 3.98)
[ INFO] [1717023122.488093492]: Call gimbal Ctrl.
[3331171.384]STATUS/1 @ rotateGimbal, L1902: Start to rotate the gimbal 0, (p,r,y) = (-15.00,0.00,0.00) mode:0 time:2.50s
[3331171.416]ERRORLOG/1 @ rotateGimbal, L1909: Rotation gimbal fail. Error code : 0xC00000000FF
[3331171.416]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Gimbal
[3331171.416]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Undefined error
[3331171.416]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Please contact <dev@dji.com> for help.
[ INFO] [1717023124.521012450]: Current gimbal 0 , angle (p,r,y) = (0.00, -120.00, 3.98)
[ INFO] [1717023124.826985037]: Current gimbal 0, angle (p,r,y) = (0.00, -120.00, 3.98)
[ INFO] [1717023124.827192144]: Call gimbal Ctrl.
[3331173.723]STATUS/1 @ rotateGimbal, L1902: Start to rotate the gimbal 0, (p,r,y) = (-30.00,0.00,0.00) mode:0 time:1.50s
[3331173.757]ERRORLOG/1 @ rotateGimbal, L1909: Rotation gimbal fail. Error code : 0xC00000000FF
[3331173.757]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Gimbal
[3331173.757]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Undefined error
[3331173.757]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Please contact <dev@dji.com> for help.
[ INFO] [1717023126.861931926]: Current gimbal 0 , angle (p,r,y) = (0.00, -119.90, 3.98)
[ INFO] [1717023135.693332636]: Current gimbal 0, angle (p,r,y) = (0.00, -119.90, 3.98)
[ INFO] [1717023135.693643712]: Call gimbal Ctrl.
[3331184.589]STATUS/1 @ rotateGimbal, L1902: Start to rotate the gimbal 0, (p,r,y) = (-50.00,0.00,0.00) mode:0 time:1.00s
[3331184.616]ERRORLOG/1 @ rotateGimbal, L1909: Rotation gimbal fail. Error code : 0xC00000000FF
[3331184.617]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Gimbal
[3331184.617]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Undefined error
[3331184.617]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Please contact <dev@dji.com> for help.
[ INFO] [1717023137.721396312]: Current gimbal 0 , angle (p,r,y) = (0.00, -120.00, 3.98)
[ INFO] [1717023138.028035182]: Current gimbal 0, angle (p,r,y) = (0.00, -120.00, 3.98)
[... cut ...]
[3331217.996]ERRORLOG/1 @ printErrorCodeMsg, L938: >>>>Error module   : Gimbal
[3331217.996]ERRORLOG/1 @ printErrorCodeMsg, L939: >>>>Error message  : Undefined error
[3331217.996]ERRORLOG/1 @ printErrorCodeMsg, L940: >>>>Error solution : Please contact <dev@dji.com> for help.
[ INFO] [1717023171.101269616]: Current gimbal 0 , angle (p,r,y) = (0.00, -119.90, 3.98)
[ INFO] [1717023171.407178976]: Current gimbal 0, angle (p,r,y) = (0.00, -119.90, 3.98)
[ INFO] [1717023171.407429412]: Call gimbal Ctrl.

dji-dev, 2024-05-30:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/107245'>Zendesk ticket #107245</a>:<br><br>Dear Erfan-Dilfanian,

Greetings. Thank you for reaching out to DJI - Da-Jiang Innovations.

There is an issue with the rotation gimbal, displaying error code: 0xC00000000FF.
This error needs to be investigated in two main areas:

1. Ensure that the specified gimbal port matches the actual physical interface of the gimbal mount.
2. Check and ensure that both A USB connections are established and communication is functioning properly.

We appreciate your support for DJI products! Wishing you all the best!

Best Regards,

DJI - Da-Jiang Innovations SDK Technical Support
<br><p><em><small>°°°</small></em></p>