dji-sdk/Onboard-SDK-ROS #538: What are the commands being sent from RC or being obtained from \dji_osdk_ros\rc topic?

spacifyy, 2024-11-12:
Hi there,

I have a question. I have a HITL setup of DJI M300, and I am saving data by recording these topics using rosbag: dji_osdk_ros/attitude, dji_osdk_ros/imu, dji_osdk_ros/local_position and dji_osdk_ros/rc.

My RC is in S mode; I am confused about what exact commands are being sent from the RC sticks and what is being saved in dji_osdk_ros/rc. I know this is scaled to -1 to 1, but are these roll pitch yaw angles commands being sent from the RC sticks or is it something else? And is the throttle thrust or z-position command?

How do you do the appropriate scaling to visualize the data or make system identification, let's suppose?

dji-dev, 2024-11-13:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/121730'>Zendesk ticket #121730</a>:<br><br>Dear Developer,

Hello, thank you for contacting DJI.

Thank you for your patience. First of all, the data you received in the range of -1 to 1 is the RC input, including the throttle range. Perhaps you can refer to:
dji_f32_t pitch; /*!< down = -0.999, middle = 0.000, up = 0.999 */
dji_f32_t roll; /*!< left = -0.999, middle = 0.000, right=0.999 */
dji_f32_t yaw; /*!< left = -0.999, middle = 0.000, right=0.999 */
dji_f32_t throttle; /*!< down = -0.999, middle = 0.000, up = 0.999 */

Do you want to combine these values with the actual flight speed of the drone for display?

Thank you for your support of DJI products! Wish you all the best!

Best Regards,
DJI SDK Technical Support
<br><p><em><small>°°°</small></em></p>

spacifyy, 2024-11-13:
Hi I understand about the commands and the scale. What I mean to ask is, when I give the pitch stick, that command is converted into a pitch angle command? pitch rate or forward velocity? If it is a pitch angle command, what factor is being used to scale [-1 1] to the appropriate pitch angle command being used by the attitude controller?

Similar to all four channels, like for throttle, what exactly does it mean? is it converted into thrust, ascent velocity, or vertical position command? And then yaw rate? is the command yaw rate or yaw angle? again what factor does it have?

I have to do system identification, I have the commands from rc, attitude and local positions, and I have to derive the following transfer functions:

1. Roll Angle Command to Actual Roll Angle
2. Pitch Angle Command to Actual Pitch Angle
3. Thrust Command or Z Position Command to Actual Z Position

Now my question is, how do I scale up the rc input data? What factor should I multiply with the data so that I could use input/output data for the system identification?

dji-dev, 2024-11-14:
Agent comment from Leon in <a href='https://djisdksupport.zendesk.com/agent/tickets/121730'>Zendesk ticket #121730</a>:<br><br>Dear developer,

Hello, thank you for contacting DJI.

Thank you for your further supplement. Sorry, I may not have fully understood your meaning before. In fact, you want to get the mapping relationship. For example, when the joystick value is 0.5, what is the corresponding thrust given to the aircraft? Is this correct?
Regarding this issue, it may involve flight control logic. It is not yet certain whether it can be made public. However, we will try our best to help you confirm it.

In addition, we noticed that you sent emails and messages through multiple channels. To avoid omissions and confusion in communication, we recommend that you communicate in the same ticket.

Thank you for your support for DJI products! Wish you all the best!

Best Regards,
DJI SDK Technical Support
<br><p><em><small>°°°</small></em></p>