dji-sdk/Onboard-SDK-ROS #541: Python SDK support for Dji flip

YESAndy, 2025-02-17:
Hi, is there any way to build a python sdk wrapper for Dji flip?

dji-dev, 2025-02-18:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/129473'>Zendesk ticket #129473</a>:<br><br>I regret to inform you that the DJI Flip does not support SDK development.
<br><p><em><small>°°°</small></em></p>

danielavdalian, 2025-02-24:
and the DJI Flip will have a sdk comming soon?

dji-dev, 2025-02-25:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/129473'>Zendesk ticket #129473</a>:<br><br>At present, there are no plans in place.
<br><p><em><small>°°°</small></em></p>

jeancode, 2025-04-24:
necesitamos el sdk de dji flip pleace

AppStudioLB, 2025-08-01:
No updates?

danielavdalian, 2025-12-06:
Hi! Why new models are receiving sdk and dji flip no?